mdl29/donkeycarLPH #10: [Organisation] Imagination des batteries à acheter / modulateurs

yannis-mlgrn, 2022-03-05:


Benvii, 2022-04-03:
Montage finalisé par Arnaud, testé ce soir, pour rappel on a donc alim de la PI et Moteur sur une NiMH 7.2V (j'ai testé avec la OPV 5 B, 7.2V) chargé juste avant, observations :
- 1 probable faux contact lorsque le PC bouge, reboot de la Pi
- après ~10min de pilotage la voiture est très "moule", la mesure sur la NiMH donne 6,6V, c'est trop faible

Refaire le test avec une batterie pas KO.

Benvii, 2022-04-09:
Test d'énergie à faire cette après midi (09/04) :
- Option 1 : KO à RPI consomme trop, test fait par Benjamin déjà
- Option 2 : à tester, faire tourner la voiture au max jusqu'à perte de pêche ou arrêt moteurs, l'autonomie théoriques serait de 50min
- Option 3 : à tester avec les batteries nimh fournies avec les voitrues 1 100 / 1 200 mah, autonomie théorique de < 20 min
- Option 4 : à tester avec la batterie li-ion de Stéphane (2 600 mah), autonomie théoriques ~25min

Faire un tableau.

Benvii, 2022-04-24:
Nous sommes parti sur l'option 4 finalement qui a fait ses preuves, autonomie de 1h30 par pack.

Maj du fichier faite il est sur le cloud LPH, reste à passer commande peut-être pour 2-3 chargeurs USB (+ câble USB vers micro USB).

yannis-mlgrn, 2022-06-12:
Issue fixed ?

Benvii, 2022-06-12:
Fixed we have everything.

closed 2022-06-12 by Benvii