mdl29/donkeycarLPH #8: [Soft] Faire la backup de la carte SD

yannis-mlgrn, 2022-03-05:


Benvii, 2022-03-05:
Le backup est fait il est sur mon poste `/home/benjamin/Documents/MDL/LPH/DonkeyCarSDBackup/2022-03-05_dababycar-debutSeanceLPH.img`, fingerprint sha256 `COMPUTE IN PROGRESS`

Benvii, 2022-03-13:
Je réouvre ce ticket car il nous faut une procédure de backup scripté ou documentée.

J'ai commencé à faire des tests avec [PiShrink](https://github.com/Drewsif/PiShrink), pour l'instant je peux confirmer que le mode (sans le cleaning des log & co, option `-p` marche).

Todo : 
- [ ] Documenter une procédure de sauvegarde (éventuellement scriptée), qui réduit l'image et la compresse
- [ ] Héberger les backup dans le cloud (bucket S3 AWS par exemple)

commit a9a6f184 (2022-04-10): :memo: [backup] fixes #8 documentation for backup procedure
commit a50ed371 (2022-04-10): Merge pull request #20 from mdl29/8-software-faire-la-backup-de-la-carte-sd

:memo: [backup] fixes #8 documentation for backup procedure
closed 2022-04-10 by Benvii