mdl29/donkeycarLPH #13: [Backup] Script pour trouver le meilleur bs (ibs / obs)

Benvii, 2022-03-19:
Basé sur cet article : http://blog.tdg5.com/tuning-dd-block-size/

- [ ]  Voir pour écrire un script qui devrai écrire un petit fichier sur la carte, l'utiliser pour calculer les vitesse de transfert depuis la SD vers le PC avec différentes block size et mettre en avant la meilleure.
- [ ] Écrire un second script pour la restoration d'un backup qui va calculer le meilleur bs également.

Adapter le documentation du backup.

Benvii, 2022-04-01:
Après de nombreux tests sur une carte SD, inspirés des scripts du site je vois que les valeurs changent régulièrement, c'est tout bonnement incompréhensible.

Pour aller plus loin voici d'autres approches : https://www.baeldung.com/linux/dd-optimal-blocksize

Je pense qu'il faut jouer sur les block size des 2 FS, donc ibs et obs pour trouver la bonne combinaisons et très certainement étudier la taille des caches/buffer également.

yannis-mlgrn, 2022-04-18:
Now we use clean images with packer. We will still need this script to make a backup?

viandoxdev, 2022-04-18:
Considering that a full dd takes about 5 min I don't think this is really necessary anymore, although faster speeds would certainly be appreciated.

closed 2022-06-11 by viandoxdev