mdl29/donkeycarLPH #18: anti-theft security for the ps4 controllers

viandoxdev, 2022-04-02:
Because otherwise people would probably just leave with the damn thing.
We will probably just modify [this](https://www.thingiverse.com/thing:4284443), remove the phone holding part, and replace it with a hook of some sort to attach a chain or a string that will link the controller to a fence or something to prevent people from just leaving with it.

viandoxdev, 2022-04-02:
@yannis-mlgrn you could first print the unaltered version (might have to look into the size of it) to make sure it can at least grip the controller and then we'll see into modifying the model.

viandoxdev, 2022-04-02:
ah merde on est en francais ici

yannis-mlgrn, 2022-04-05:
voici mon antivol pour manette de PS4 : 
- [lien thinguiverse](https://www.thingiverse.com/thing:5342314)

PR #38 (2022-04-18): Fixes #18 : Create anti-theft for ps4 controllers
Fixes #18
commit 28b65c24 (2022-04-18): Fixes #18 : Create anti-theft for ps4 controllers
commit ce9ecbe4 (2022-04-18): Merge pull request #38 from yannis-mlgrn/18-anti_theft_ps4_controller

Fixes #18 : Create anti-theft for ps4 controllers
closed 2022-04-18 by yannis-mlgrn