mdl29/donkeycarLPH #19: [Car4] Contruire la voiture 4

Benvii, 2022-04-03:
On a pris une 4ème voiture, basé sur un chassie différent et aussi à notre mauvaise surprise des composants différents, ce qu'il reste à faire pour qu'il puisse tourner :

- [ ] Remplacement du composant ICS+ESC (+BEC) cad récepteur RF + variateur par un variateur où on pourra piloter le moteur en PWM, j'ai commandé un `QuicRun 1625 Brushed ESC 25A` chez rc-race-shop.de
- [ ] Conception / adaptation de la pièce de support pour la RPi la fixation ne peux pas être la même et impression des pièces
- [ ] Équiper d'une Pi4 + PiCam
- [ ] Souder les différents cables (bloquer en attente du composant)

yannis-mlgrn, 2022-04-05:
⚠️ The right steering arm is broken
- ref : 6260 steering arms 
zdhobby

yannis-mlgrn, 2022-04-09:
🪛 ~ Réparation de la voiture N°4 
- pièce recollée ( c'est résistant ) 
Prévoir qu'elle puisse cassée dans le futur

yannis-mlgrn, 2022-04-18:
We can close

closed 2022-04-18 by yannis-mlgrn