mdl29/donkeycarLPH #27: [Ansible] Role "base", basic raspberry pi configurations

Benvii, 2022-04-12:
Create a base role, this role will be used for generic raspberry pi configuration stuff not directly related to DonkeyCar projet.

This role should : 

- [ ] Enable SSH, by creating /boot/ssh file, using Ansible [file module with state: touch](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/file_module.html)
- [ ] Configure wifi, based on a template file that will be copied to `/etc/wpa_supplicant/wpa_supplicant.conf`. Parameters (country code, wifi ssid and psk) should use variables defined in group_vars/all
- [ ] Perform update and upgrade using the [Ansible apt module](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/apt_module.html#ansible-collections-ansible-builtin-apt-module) see in the exemple task `Run the equivalent of "apt-get update" as a separate step`, for upgrade use `upgrade: safe` option see exemples in documentation
- [ ] Change password for pi user, with password defined as variable, you might take a [look at this article](https://jianjye.medium.com/how-to-update-user-password-with-ansible-f971f41a3b3e) or simply use command `chpasswd` (see [how raspi-config does the job](https://github.com/raspberrypi-ui/rc_gui/blob/master/src/rc_gui.c#L132))
- [ ] Enable I2C [look at here](https://github.com/raspberrypi-ui/rc_gui/blob/master/src/rc_gui.c#L81)
- [ ] Enable camera

Take care to use named tasks and templates.

That's all :)

yannis-mlgrn, 2022-04-13:
`%d` == 1

Benvii, 2022-04-15:
You also need to install pigpio :
```bash
sudo apt update && sudo apt install pigpio python3-pigpio
sudo systemctl start pigpiod
```

commit b2fd27cf (2022-04-13): fixes #27 : create rspi-config role
PR #32 (2022-04-16): [Ansible] Fixes #27 #28 # 29 #30 rpi config / donkeycar / openCV  role 
Fixes #27 #28 #29 #30

Add rpi-config related stuff role.
commit f439365e (2022-04-16): [Ansible] Fixes #27 #28 # 29 #30 rpi config / donkeycar / openCV  role  (#32)

* work on rapsi-config

* fixes #27 : create rspi-config role

* remove a mistake

* remove new line at end of files

* remove space

* Fixes somes error with raspi-config roles and resolves #28

* change dev host hostname

* change commentary

* Fixes #29 : OpenCV roles

* add gpio service

* First draft of my car roles

* :bug: [Ansible] Fixes wrong open-cv package name to python3-opencv (as written in donkeycar's documentation)

* :bug: [Ansible] Fixes task doing nothing to notify pigpio start

* :bug: [Ansible]  [... cut]
closed 2022-04-16 by Benvii