mdl29/donkeycarLPH #28: [Ansible] Role "donkeycar"

Benvii, 2022-04-12:
**Needs #26 to be merged before.**

Create a new role "donkeycar", will represent a basic donkeycar installation (without optionnal stuff such as OpenCV..). You will find all informations in donkeycar documentation website : https://docs.donkeycar.com/guide/robot_sbc/setup_raspberry_pi/ 

This role should :

- [ ] Step 8: Install Dependencies, using Ansible APT module, you can install multiple modules with : https://stackoverflow.com/a/63668940
- [ ] Step 10: Setup Virtual Env, take a look at how we have [created virtualenv in Open Path View](https://github.com/OpenPathView/OPV_Ansible/blob/master/roles/base/tasks/opv.yml#L28), don't forget to add source line to bashrc using [lineinfile module](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/lineinfile_module.html). If you don't use the linInFile module line might be added multiple times if we re-excute the book
- [ ] Step 11: Install Donkeycar Python Code, clone repository (using [git module](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/git_module.html)), use [file module](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/file_module.html) to create directories and [pip module](https://docs.ansible.com/ansible/latest/collections/ansible/builtin/pip_module.html) for pip commands

Good luck :)

yannis-mlgrn, 2022-04-14:
It necessary to put the step 10 into the middle of the step 11

yannis-mlgrn, 2022-04-14:
pip install -e : https://stackoverflow.com/questions/24449134/how-to-run-python-setup-py-develop-command-inside-virtualenv-using-ansible

commit 1fe44bdc (2022-04-14): Fixes somes error with raspi-config roles and resolves #28
commit f439365e (2022-04-16): [Ansible] Fixes #27 #28 # 29 #30 rpi config / donkeycar / openCV  role  (#32)

* work on rapsi-config

* fixes #27 : create rspi-config role

* remove a mistake

* remove new line at end of files

* remove space

* Fixes somes error with raspi-config roles and resolves #28

* change dev host hostname

* change commentary

* Fixes #29 : OpenCV roles

* add gpio service

* First draft of my car roles

* :bug: [Ansible] Fixes wrong open-cv package name to python3-opencv (as written in donkeycar's documentation)

* :bug: [Ansible] Fixes task doing nothing to notify pigpio start

* :bug: [Ansible]  [... cut]
closed 2022-04-16 by Benvii