mdl29/donkeycarLPH #33: [Ansible] Role ps4-controller

Benvii, 2022-04-13:
Create a role to install ds4drv service to handle only 1 controller at a time by car.

This role should :
- [ ] Install ds4drv using Ansible git module from : https://github.com/Benvii/ds4drv/tree/193_limit_simultaneous_controllers  and their service file
- [ ] Their own LED color, will be put in a configuration file based on the ethernet MAC address, read it from a file of find a wait to deploy it in ds4drv configuration at the right time, we might need to create a service to deploy "car specific" configuration

Minemew2, 2022-04-14:
To create ds4drv service, first create a folder named donkeycar in /etc/. In this folder, create a txt file named controller_color.txt. In this file, write the choosen color for your controller, in hexadecimal format (ie : af9666).

Minemew2, 2022-04-14:
To handle devices with ds4drv, it is necessary to do the following command lines :
sudo wget https://raw.githubusercontent.com/chrippa/ds4drv/master/udev/50-ds4drv.rules -O /etc/udev/rules.d/50-ds4drv.rules
sudo udevadm control --reload-rules
sudo udevadm trigger

Minemew2, 2022-04-14:
Then, clone the following git repository in /home/pi : https://github.com/Benvii/ds4drv.git

Minemew2, 2022-04-14:
You also need to checkout to this branch : 193_limit_simultaneous_controllers

Minemew2, 2022-04-14:
In the repo file, install  ds4drv python files : $ python setup.py install

Minemew2, 2022-04-14:
Now create the service. In /etc/systemd/system/, create a txt file name ds4drv.service. Put the following code in this file : 
[Unit]
Description=ds4drv daemon
Requires=bluetooth.service
After=bluetooth.service

[Service]
EnvironmentFile=/etc/donkeycar/ds4drv.env
ExecStart=sudo /home/pi/env/bin/ds4drv --controller-limit 1 --led ${controller_color}
Restart=on-abort

[Install]
WantedBy=bluetooth.target

PR #39 (2022-04-19): Fixes #33 #43 : Create ps4-controller role
Fixes  #33 #43
commit 32843a09 (2022-04-19): Fixes #33 #43 : Create ps4-controller role (#39)

* Fixes #33 : Create ps4-controller role

* add color var

* fixed 3 first errors

* Might be the final commit

* remove sudo for run setup.py

* removed wrong python path for run setup.py

* Another bad path resolved

* Uncomment roles in donkeycar.yml

* fixing ds4drv install using pip and git URL

* fixes some review

* Fixes all @benvii reviews

* add modif

* add sleep in ds4drv.service

* Fix line issue after rebasing

* remove path env typoq

* fixes typos variable in ds4drv.env

* modifie the dhcpd5 service

* Fixing some issues after r [... cut]
closed 2022-04-19 by Benvii