mdl29/donkeycarLPH #34: [Ansible] wpa_suicant service not restarted

yannis-mlgrn, 2022-04-14:
The animator after use ansible to install and configure packages for run donkeycar need to restart the wpa_suplicant service for connect the rpi to the wanted wifi.But today we can't put that in the ansible recipe because it shutdown the connection internet during few second and disconnect the ssh connection. In the future for avoid to restart the car after change the `/etc/wpa_suplicant/wpa_suplicant.conf`
we want to restart the service after have finished the ansible recipe for avoid this disconnection. 
See this issue : https://stackoverflow.com/questions/39994218/ansible-to-restart-network-service

yannis-mlgrn, 2022-04-25:
Issue who don't have any importance now 
@Benvii ??

Benvii, 2022-04-26:
Yes it's working now and fixed.

closed 2022-04-26 by Benvii