mdl29/donkeycarLPH #44: Start manage.py drive at startup

Minemew2, 2022-04-17:
Start manage.py drive at startup

Minemew2, 2022-04-17:
donkeycar.service:
```
[Unit]
Description=Start donkeycar drive on boot, to be able to drive directly the car
Requires=ds4drv.service
Requires=pigpiod.service
After=ds4drv.service
After=pigpiod.service
[Service]
User=pi
ExecStartPre=/bin/sleep 5
ExecStart=/home/pi/env/bin/python /home/pi/mycar/manage.py drive --js
Restart=on-abort

[Install]
WantedBy=multi-user.target

```

Minemew2, 2022-04-17:
ds4drv.service:
```
[Unit]
Description=ds4drv daemon
Requires=bluetooth.service
After=bluetooth.service

[Service]
EnvironmentFile=/etc/donkeycar/ds4drv.env
ExecStartPre=/bin/sleep 5
ExecStart=/home/pi/env/bin/ds4drv --controller-limit 1 --led ${controller_color} --config /etc/donkeycar/ds4drv.conf --bindings donkeycar
Restart=on-abort

[Install]
WantedBy=bluetooth.target
```

yannis-mlgrn, 2022-04-18:
Why

PR #46 (2022-04-24): Fixes #44: Launch donkeycar.service at startup
Fixes : #44
commit a6d82727 (2022-04-24): Fixes #44: Launch donkeycar.service at startup (#46)

* Fixes #44: Launch donkeycar.service at startup

* Remove venv path typos

* install pandas and others dependencies

* enable camera and I2C

* pip with pi user

* update mycar roles

* password change : ✅
camera enabled (photo taken):  ✅
i2c (device i2c visible) : ✅

* Fix some pip not run as "pi" and adding back venv sourcing in bashrc file

* Attempt to fix permission issue

* fixes mycar handlers problem

* Reordering roles, more logically

* Permission issue fix

* Fixes #52 : Wrong myconfig file

* Fixes #50 : increase timeout and re [... cut]
closed 2022-04-24 by Benvii