mdl29/donkeycarLPH #53: [Drive] Throttle limiting for physical controller

Benvii, 2022-04-24:
# Use case

As a driver and animator I want to be able to limit the throttle (apply a limiting factor on it), not only for the web interface but for all controllers interfaces. From the feature we would have different driving mode such as kids mode (at 50%), expert mode at (100 %) ...

# Actual implementation

Throttle limitation is only available for web controllers (from the web interface), the limitation is applied before sending values to the car.

# Evolution 

We might have different approaches : 

* See if we can use calibration commands to do the limiting
* Or add a limitation factor with associated websocket commands and apply this factor on the car for all controllers

yannis-mlgrn, 2022-05-05:
## How throttle is used in donkeycar drive
```html 
<option value="0.05">5%</option>

<option value="0.1">10%</option>

<option value="0.15">15%</option>

<option value="0.2">20%</option>

<option value="0.25">25%</option>

<option value="0.3">30%</option>

<option value="0.35">35%</option>

<option value="0.4">40%</option>

<option value="0.45">45%</option>

<option value="0.5">50%</option>

<option value="0.55">55%</option>

<option value="0.6">60%</option>

<option value="0.65">65%</option>

<option value="0.7">70%</option>

<option value="0.75">75%</option>

<option value="0.8">80%</option>

<option value="0.85">85%</option>

<option value="0.9">90%</option>

<option value="1.0">100%</option>
```
## How data throttle is posted  : 
```js
    var postDrive = function(fields=[]) {

        if(fields.length === 0) {
            fields = ALL_POST_FIELDS;
        }

        let data = {}
        fields.forEach(field => {
            switch (field) {
                case 'angle': data['angle'] = state.tele.user.angle; break;
                case 'throttle': data['throttle'] = state.tele.user.throttle; break;
                case 'drive_mode': data['drive_mode'] = state.driveMode; break;
                case 'recording': data['recording'] = state.recording; break;
                default: console.log(`Unexpected post field: '${field}'`); break;
            }
        });
        if(data) {
            console.log(`Posting ${data}`);
            socket.send(JSON.stringify(data))
            updateUI()
        }
    };
```

yannis-mlgrn, 2023-01-21:
We can ajust it with ps4 controller, we can close this issue now

closed 2023-01-21 by yannis-mlgrn