mdl29/donkeycarLPH #54: [Hardware] Camera problem

yannis-mlgrn, 2022-04-24:
I have 2 errors who might be caused by a hardware error : 
```
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v4.3.13 ...
INFO:donkeycar.config:loading config file: /home/pi/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:__main__:PID: 760
cfg.CAMERA_TYPE PICAM
INFO:__main__:cfg.CAMERA_TYPE PICAM
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0xee2230 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "manage.py", line 853, in <module>
    meta=args['--meta'])
  File "manage.py", line 142, in drive
    cam = PiCamera(image_w=cfg.IMAGE_W, image_h=cfg.IMAGE_H, image_d=cfg.IMAGE_DEPTH, framerate=cfg.CAMERA_FRAMERATE, vflip=cfg.CAMERA_VFLIP, hflip=cfg.CAMERA_HFLIP)
  File "/home/pi/projects/donkeycar/parts/camera.py", line 27, in __init__
    self.camera = PiCamera() #PiCamera gets resolution (height, width)
  File "/home/pi/venv/lib/python3.7/site-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/home/pi/venv/lib/python3.7/site-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/home/pi/venv/lib/python3.7/site-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/home/pi/venv/lib/python3.7/site-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/home/pi/venv/lib/python3.7/site-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources
```
and I can't take pictures witth the rpi : 
```
(venv) pi@bluecar:~/mycar $ raspistill -o ~/img.jpg
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

```

Benvii, 2022-04-24:
Bluecar's camera was hurt during our last driving session, you should test it from a simple/standard raspbian installation and with raspi-config CLI to enable it.

We didn't have any issue before with it.

Benvii, 2022-04-24:
I also bought 2 Pi Cam V2 board, and 2 other wide angle modules, at most we will have 1 spare camera (as redcar already needs one).

yannis-mlgrn, 2022-04-25:
# IMPORTANT !!!
I have the same error with dababycar, why ? 
maybe because the donkeycar.service is launch on boot and create conflict with the camera !!!

yannis-mlgrn, 2022-04-25:
I have resolved this issue, don't run in same time two times the donkey drive and donkeycar service

closed 2022-04-25 by yannis-mlgrn