mdl29/donkeycarLPH #55: [Ansible] Shortcut to take full resolution picture (check camera quality)

Benvii, 2022-04-25:
En tant qu'animateur je souhaites pouvoir vérifier que le capteur photo est OK (focus, saletés ...).

On pourra simplement effectuer une prise de photo via un bouton de la manette à un emplacement fixe (exemple /tmp/testPhoto.jpg ) et fournir une documentation sur comment récupérer l'image (scp).

Attention pour faire un raspistill il faut stopper le service donkeycar qui utilise la caméra, prendre la photo, relancer le service. C'est pas un problème de le stopper comme un fera cette vérification en cas de doute donc quand la voiture tourne pas (probablement au démarrage).
Penser a faire une doc sur comment récupérer l'image via scp, on ne cherchera pas à faire mieux pour l'instant.

Benvii, 2022-04-26:
@yannis-mlgrn you want to take this ticket ?

yannis-mlgrn, 2022-05-01:
Of courses

PR #64 (2022-05-11): Resolves #55 :  [Ansible] Shortcut to take full resolution picture (check camera quality)
# ✨ Feature : 
- Fixes #55 . 
 I have introduce a command to allow a user to take a picture with ps4 controller for test the resolution or verify that the camera running
**The shortcut is : PS + Triangle + L2 + R2**

## ✅ Tests :  
- [x] Manual testing on dababycar
- [x] image testing

**ALL TESTS PASSED**
commit feaa6ab2 (2022-05-11): Resolves #55 :  [Ansible] Shortcut to take full resolution picture (check camera quality) (#64)

* ✨ ~ Create command for take a picture

* ⚡ ~ Resolves problems with command execution
closed 2022-05-11 by Benvii