mdl29/donkeycarLPH #57: [hardware] Don't loose any camera cap (rpi cam 2 wide angle)

Benvii, 2022-04-26:
As a donkeycar user I will find it pretty useless to have a sort of camera cap support on the 3D printed parts.

It would simply be a cone of the size of the camera cap so that we can put the cap on it when it's not on the camera.

Todo :

- [ ] Add a cone of the cap size : diameter 14 mm, height 7mm (no more needed)
- [ ] Print it on a small piece, to test it
- [ ] Then add it to donkeycar base part

yannis-mlgrn, 2022-05-03:
where is the better position for allow the user to put and re

Benvii, 2022-05-04:
We need 2 STL separate files so that we can easily print them, can you generate 2 separate files ? Thanks

Benvii, 2022-05-14:
@yannis-mlgrn can you split them into 2 separate STL so that I can print them and replace the parts ?

yannis-mlgrn, 2022-05-14:
Already done

Benvii, 2022-05-14:
Thanks it's here : https://github.com/mdl29/donkeycarLPH/blob/main/3dDesigns/camera-cover/lowPart.stl

Benvii, 2022-05-15:
Missing some screw to fix the raspberry pi, order was made they should arrive by the end of next week.

Benvii, 2022-06-08:
I'm closing it, not sure we will push it to donkeycar community

PR #66 (2022-05-04): Resolves #57 : Create camera cover holder
# ✨ Feature :
- Fixes #57 ,
 Create a camera cover holder for the donkeycar chassis to avoid to lose one
commit 8ae4601a (2022-05-04): Resolves #57 : Create camera cover holder (#66)

* ✨ ~ Create camera cover on donkeycar chassis

* 🚚 ~ Improve doc organization
closed 2022-06-08 by Benvii