mdl29/donkeycarLPH #102: Discover Donkeycar manager host using zero conf

Benvii, 2022-05-28:
We don't know what will the computer hosting donkeycar manager API and it's hostname or IP address. We might use zeroconf to discover it.

viandoxdev, 2022-06-01:
From a quick read of ubuntu's zeroconf page, this should be done using a service (i.e. the donkeycar manager host declares itself as being one) and just pick the first host that shows this capability with avahi-discover ?

Benvii, 2022-06-07:
I needed it today and made quick test using python zeroconf it pretty easy see 2 demo scripts here : 
- https://github.com/mdl29/donkeycarLPH/pull/117/commits/25d313fb16975b9bf501eb46600ba192e4a5bc98

I will add it to the backend and to my PR.

closed 2022-06-18 by Benvii