mdl29/donkeycarLPH #110: Battery life test

Benvii, 2022-06-04:
We need to log CPU load  (each 30 sec) :
https://www.howtogeek.com/devops/how-to-monitor-cpu-usage-in-linux-over-time/

Then try to generate a load X% : https://superuser.com/a/668332

MaximeLM29, 2022-06-04:
```bash
top -bn1 | grep "Cpu(s)" |            sed "s/.*, *\([0-9.]*\)%* id.*/\1/" |            awk '{print 100 - $1"%"}'
```

yannis-mlgrn, 2022-06-05:
Should be have it on dashboard donkeycar in donkeycar front. I think is useful to monitoring cpu cars...

yannis-mlgrn, 2022-06-19:
much like that : ```top -bn1 | grep "Cpu(s)" |            sed "s/.*, *\([0-9.]*\)%* id.*/\1/" |   awk '{print "cpu :",  $1"%"}' ```

yannis-mlgrn, 2022-06-19:
something like that but I can't get the percentage 
```awk 
top -bn1 | grep "Cpu(s)" | sed "s/.*, *\([0-9.]*\)%* id.*/\1/" | awk 'BEGIN { print "[",strftime("%m/%d/%Y %H:%M:%S", systime()),"]","cpu :",$0,"%"}'
```
Output : `[ 06/20/2022 00:08:53 ] cpu :  %`

yannis-mlgrn, 2022-06-19:
I should use `cron`  for log these information's and in which file i put these logs. A idea @Benvii  ?

Benvii, 2022-06-19:
Tests were done we have a battery life of ~ 3h for 10 Ah batteries.

A good way to handle it would be to have a "countdown" and at each car restart ask if the battery was change then display battery life on the management.. we did anticipate it and I'm not sure we will have the time to develop it

yannis-mlgrn, 2022-06-19:
so close it  ?

yannis-mlgrn, 2022-07-04:
@Benvii ?

Benvii, 2022-07-05:
No we will simply keep the idea and do it for a next version

Benvii, 2023-05-08:
New battery made

closed 2023-05-08 by Benvii