mdl29/donkeycarLPH #184: [Front] extend job duration

viandoxdev, 2022-07-09:
Add a way to extend a job's duration in the dashboard, for people whose car broke on their turn (after the timer has been started). This would be much faster (and less annoying) than having to create a new job and put it in first.

yannis-mlgrn, 2022-07-09:
it's impossible for @Benvii , the good way according to his is the same that our answer

yannis-mlgrn, 2022-07-13:
I can create a button for replace the pause button who it always disable. I would create a button named reload who create the same job, put it in the first place and cancel the one who don't work.
ok, @viandoxdev  ?

Benvii, 2022-09-17:
Job as to be created with the same worker assigned and before the previous job is ended (so that we don't have any other job assigned to this same worker that would executed before the job we want).

PR #190 (2022-09-24): [ Front ] Add reload button in dashboard
Fixes #184,
I've added a reload button for give the possibility at a admin to restart a job if have a problem during the race with the car or anything else. Moreover, I've chosen this solution because we can't extend the job duration.
closed 2022-09-24 by Benvii