mdl29/donkeycarLPH #218: [Front] AI driving job

Benvii, 2022-09-26:
A new job is introduced by #217 .

It can be created using : 
```json
{
        "worker_type": "CAR",
        "name": "AI_ASSISTED",
        "parameters": "{\"model_remote_archive\": \"models/2022-09-25_20:08_p6_j19.tar.gz\", \"drive_time\": 120}",
        "state": "WAITING",
        "worker_id": null,
        "player_id": 3,
        "next_job_details": null,
        "screen_msg": null,
        "screen_msg_display": false,
        "job_id": 5,
        "created_at": "2022-09-25T20:00:16.796859+00:00",
        "rank": 7000,
        "fail_details": ""
}
```

With [2022-09-25_20:08_p6_j19.tar.gz](https://github.com/mdl29/donkeycarLPH/files/9649763/2022-09-25_20.08_p6_j19.tar.

User will be told to press X to start driving the model then the car will automatically moves. This job still handle races.

Todo : 

- [ ] Remove the sentence telling the user to move to start driving (it doesn't make sens for this job)
- [ ] Find a new icon like "🧠"

yannis-mlgrn, 2022-09-29:
but when we create it  ?

Benvii, 2022-09-29:
This job is created by the car (at the end of the recording job), you don't have to create it. It will be created automatically

commit 1b998371 (2022-10-02): [Front] fix #218 (#221)

* Fixes #218

* add icon and indicator for ai jobs

* Add missing file
closed 2022-10-02 by Benvii