mdl29/donkeycarLPH #261: UX : Visual way to explain how training works

Benvii, 2023-05-01:
We just have 1 sentences explaining that the training is in progress, in fact we would really need more than that. We though of several little animation that would explain it while the training is in progress (or waiting for an AI runner).

Illustrations here : https://photos.app.goo.gl/xQVnCwuDEJDePayx8 

We will have several stages : 

1. Explain that the collected data set is split into 2 portions (during max 40 sec)
2. Show the AI as a student in a class room learning the notion of turning right and left with a teacher (during 30-40 sec)
3. Explain how the model is evaluated (it's precision, recall and so on), with the AI being tested on some images that we keep out of it's training

First illustration made by @Nils-Abjorn are available here (we should make a PR) : 

1. [image]
2. [image]
3. [image]

Nils-Abjorn, 2023-05-06:
[image]

Nils-Abjorn, 2023-05-06:
[image]

PR #276 (2023-06-25): Resolves #261: visual way to explain
Fixes #261 
- [x] Fixe css
commit 3641cb12 (2023-06-25): Resolves #261: visual way to explain (#276)

* ⚡ add UI component

* ✨ Improve css and js script

* 🩹 Fix css

* ✨ Improve responsive and size

* forget to change time

* 💩 modify the car code for add screen_msg

* put test under UI

* 💩 refracto code

* done ?

* V1 - fix bugs

* final change robotcup fixes

* last titou fix

* ♻  Remove UI-train in car ansible

* Revert "♻  Remove UI-train in car ansible"

This reverts commit 959880dbe17f5d1f78c214bfe8251c0249351cab.

* ♻ Remove 'UI-train' and others change are PEP8 linter

* fix benvi review

* UI fixes

* :pencil2: The final touch

------- [... cut]
closed 2023-06-25 by Benvii